Pioreactor/hardware #20: Pinout diagram of HAT

CamDavidsonPilon, 2021-06-21:
https://github.com/j0ono0/pinout/tree/0.0.10

VitorFrost, 2022-01-28:
Can you post here at list the circuit diagram? anything? 
I just want to understand a little more how things are connected, and what sensors are used.
I do believe I am using a different temperature sensor. I also would like to know what port is controlling the leds.

VitorFrost, 2022-02-01:
Thanks for the info, that is enough for me.  I am using the `pin 4` for 1-wire communication, to connect a temperature sensor (Dallas DS16B20).
I have chosen it because I am housing the sensor inside an straw tube, that is inserted in the reactor! So I restricted the sensor to the encapsulation TO-92, to be digital and works with 3,3V .

CamDavidsonPilon, 2022-02-01:
> I have chosen it because I am housing the sensor inside an stainless straw tube, that is inserted in the reactor

That's a cool idea!

closed 2023-03-15 by CamDavidsonPilon