Pioreactor/hardware #25: use ads1015

CamDavidsonPilon, 2021-07-14:


VitorFrost, 2022-01-22:
I am planning to use this to read an ADS1115 to read an pH sensor for Arduino.
Do you have any suggestion where to start it?

VitorFrost, 2022-01-22:
Does this helps you?
https://github.com/msurguy/processing-mqtt-sensor-demo

CamDavidsonPilon, 2022-01-22:
Our current development HATs use the ADS1115, but we plan to use 1015 for production. 

> I am planning to use this to read an ADS1115 to read an pH sensor for Arduino.

Is it correct that `Raspberry Pi <-> Arduino <-> ADS1115`? What is the purpose of the Arduino? 

Currently there aren't any Arduino tools in Pioreactor.

VitorFrost, 2022-01-22:
> Our current development HATs use the ADS1115, but we plan to use 1015 for production.
>
>
> Is it correct that `Raspberry Pi <-> Arduino <-> ADS1115`? What is the purpose of the Arduino?
>
> Currently [...]

Actually, no. I am planning to use this sensor/module : https://www.e-tinkers.com/2019/11/measure-ph-with-a-low-cost-arduino-ph-sensor-board/
There are 3,3V of it, but they are very rare and _expensiver_ . 
I am planning to use like this:
`Raspberry Pi <-> Logic level shifter <-> ADS1x15`

VitorFrost, 2022-01-22:
Actually, pH sensing can be a must for bioreactors! Sometimes, it is more precise to control growth media addition by pH level, then by turbidimetry.

CamDavidsonPilon, 2022-01-22:
Your setup makes sense to me. And it's over i2c, so like:

```
             i2c                     i2c      voltage
Raspberry Pi <-> Logic level shifter <-> ADS1x15 <- pH Sensor
```

Here's a really quick sketch of what the Pioreactor software might look like to read from the ADS1115:

```python
# -*- coding: utf-8 -*-
# filename: ~/.pioreactor/plugins/sense_ph.py
import click

from pioreactor.whoami import get_unit_name, get_latest_experiment_name
from pioreactor.config import config
from pioreactor.background_jobs.base import BackgroundJob
from pioreactor.utils import local_persistant_storage
from pioreactor.utils.timing import RepeatedTimer
from pioreactor.hardware import SCL, SDA

from adafruit_ads1x15.analog_in import AnalogIn 
from adafruit_ads1x15.ads1115 import ADS1115 as ADS
from busio import I2C 

class PHSensor(BackgroundJob):

    published_settings = {
        "pH": {"datatype": "float", "settable": False},
    }

    def __init__(self, sample_rate, unit, experiment):

        self.ads = ADS(I2C(SCL, SDA), data_rate=128, gain=1)
        self.analog_in = AnalogIn(self.ads, 0)
        self.pH = None

        self.read_ph_timer = RepeatedTimer(
            interval=1/sample_rate, # convert to seconds
            function=self.read_ph,
            run_immediately=True,
        ).start()

    def read_ph(self):
        raw = self.analog_in.voltage

        with local_persistant_storage("ph_calibration") as cache:
            # this published to MQTT
            self.pH = raw_to_calibrated(raw, cache) #raw_to_calibrated to be defined... 

    def on_ready_to_sleeping(self) -> None:
        self.read_ph_timer.pause()

    def on_sleeping_to_ready(self) -> None:
        self.read_ph_timer.unpause()

    def on_disconnect(self) -> None:
        self.read_ph_timer.cancel()

@click.command(name="sense_ph")
def click_sense_ph():
    """
    Start the pH sensor
    """

    job = PHSensor(
        sample_rate=config.getfloat(),
        unit=get_unit_name(),
        experiment=get_latest_experiment_name(),
    )
    job.block_until_disconnected()

```

and it can be run with `pio run sense_ph`

VitorFrost, 2022-01-26:
Thanks for the file `sense_ph.py`
Just to clarify, if I am doing it wright:
I need to put this file in the folder `/home/pi/.pioreactor/plugins`
then run the command `pio run sense_ph` ?
Tried it, and got the error (tried to reboot it too):
```
pi@pioreactor:~ $ pio run sense_ph
Usage: pio run [OPTIONS] COMMAND [ARGS]...
Try 'pio run --help' for help.

Error: No such command 'sense_ph'.
```

Should the Web interface show anything? My `/plugins` page was empty too:
[image]

I have also installed de pip adafruit-ads1x15 library (https://learn.adafruit.com/raspberry-pi-analog-to-digital-converters/ads1015-slash-ads1115) and it worked fine.

Then I tried: 
```
pio update --app
pio update --ui
```
NOTE:  not able to do just` pio update`, it asked me what to update.

Nothing YET. Then, after another reboot, It showed on the Web UI, and also on the pio list-plugins.
Beside this, when I try to run `pio run sense_ph` got:
```
Usage: pio run [OPTIONS] COMMAND [ARGS]...
Try 'pio run --help' for help.

Error: No such command 'sense_ph'.
```
And the webpage plugin link points to `/Unknown` (Can a plugin point to other pages outside pioreacors default plugins?)
[image]

Any suggestion? Is the problem related to not having a proper Pioreactor HAT? If so, can I bypass any HAT self-check?

CamDavidsonPilon, 2022-01-27:
I don't think it's HAT related. 

Hm, you shouldn't need to reboot so often. Things should "just work". If you run `pio run --help`, what shows up? I'm guessing:
```
Usage: pio run [OPTIONS] COMMAND [ARGS]...

Options:
  --help  Show this message and exit.

Commands:
  backup_database         (leader only) Backup db to workers.
  export_experiment_data  (leader only) Export tables from db.

```

If so, it's because your Pioreactor isn't set as a worker. 

----

Ah! Looking back, this is an issue with first_boot again! I actually didn't give you the full list commands to run in [my comment here](https://github.com/Pioreactor/pioreactor/issues/238#issuecomment-1018824136). 

You'll need to run the rest: 
```
touch ~/home/pi/config_$(hostname).ini
printf "# Any settings here are specific to $(hostname), and override the settings in config.ini\n\n" >> /home/pi/.pioreactor/config_$(hostname).ini
cp ~/home/pi/config_$(hostname).ini $PIO_DIR/unit_config.ini
crudini --set ~/home/pi/config.ini network.inventory $(hostname) 1
```

The above edits your configuration file to make your Pioreactor a worker - _then_ you should be able to run `pio run sense_ph`. 

----

The script I gave you is just a demo, but there were a few silly bugs in it. Try [this updated version](https://gist.github.com/CamDavidsonPilon/e6c805593d718e0297641d8e47d903c0), plus that provides some metadata so information on the /plugins page is filled in.

VitorFrost, 2022-01-27:
@CamDavidsonPilon Thanks for the diagnosis!
Yes, you was wright about `pio run --help`
But the command above has an error! you have added a `~/home/pi/` where should be only `~/`
I got this kind of error:
```
$ touch ~/home/pi/config_$(hostname).ini
touch: cannot touch '/home/pi/home/pi/config_pioreactor.ini': No such file or directory

```

Now I can finally see the reactor on the page `/pioreactors`
[image]

And only after a reboot I was able to run `pio run sense_ph`
Thanks a lot for the work on the pH sensor!
Now I have to figure out how to proper program in python and fix the `name 'raw_to_calibrated' is not defined`  LOL

closed 2023-03-15 by CamDavidsonPilon